dreed47/WifiMQTTManager #1: giving error for **getting chip id**

apanasara, 2018-12-17:
throwing error for **getting chip id** while testing on *Node MCU*

dreed47, 2018-12-17:
what specific board do you have?  What do you have the "board" set to in Arduino IDE?

apanasara, 2018-12-17:
**_Example Code (I used)_**
`
```
#include "secrets.h"
#include <WiFiMQTTManager.h>

// Button that will put device into Access Point mode to allow for re-entering WiFi and MQTT settings
#define RESET_BUTTON 0

WiFiMQTTManager wmm(RESET_BUTTON, AP_PASSWORD);  // AP_PASSWORD is defined in the secrets.h file

void setup() {
  Serial.begin(9600);
  Serial.println(F("WiFiMQTTManager Basic Example"));
  // set debug to true to get verbose logging
  // wm.wm.setDebugOutput(true);
  // most likely need to format FS but only on first use
  // wm.formatFS = true;
  // optional - define the function that will subscribe to topics if needed
  wmm.subscribeTo = subscribeTo;
  // required - allow WiFiMQTTManager to do it's setup
  wmm.setup(__SKETCH_NAME__);
  // optional - define a callback to handle incoming messages from MQTT
  wmm.client->setCallback(subscriptionCallback);
  
}

void loop() {
  // required - allow WiFiMQTTManager to check for new MQTT messages, 
  // check for reset button push, and reconnect to MQTT if necessary 
  wmm.loop();

  // optional - example of publishing to MQTT a sensor reading once a 1 minute
  long now = millis();
  if (now - wmm.lastMsg > 60000) {
    wmm.lastMsg = now;
    float temperature = 70; // read sensor here
    Serial.print("Temperature: ");
    Serial.println(temperature);
    char topic[100];
    snprintf(topic, sizeof(topic), "%s%s%s", "sensor/", wmm.deviceId, "/temperature");
    wmm.client->publish(topic, String(temperature).c_str(), true);
  }

}

// optional function to subscribe to MQTT topics
void subscribeTo() {
  Serial.println("subscribing to some topics...");  
  // subscribe to some topic(s)
  char topic[100];
  snprintf(topic, sizeof(topic), "%s%s%s", "switch/", wmm.deviceId, "/led1/output");
  wmm.client->subscribe(topic);
}

// optional function to process MQTT subscribed to topics coming in
void subscriptionCallback(char* topic, byte* message, unsigned int length) {
  Serial.print("Message arrived on topic: ");
  Serial.print(topic);
  Serial.print(". Message: ");
  String messageTemp;
  
  for (int i = 0; i < length; i++) {
    Serial.print((char)message[i]);
    messageTemp += (char)message[i];
  }
  Serial.println();

  //if (String(topic) == "switch/esp1234/led1/output") {
  //  Serial.print("Changing led1 output to ");
[... 245 lines cut ...]
C:\Users\Amit Suani\Documents\Arduino\libraries\WiFiMQTTManager_Library\src\WiFiMQTTManager.cpp:282:8: error: 'ArduinoJson::JsonObject' has no member named 'printTo'

   root.printTo(messageBuffer, sizeof(messageBuffer));

        ^

C:\Users\Amit Suani\Documents\Arduino\libraries\WiFiMQTTManager_Library\src\WiFiMQTTManager.cpp:286:8: error: 'ArduinoJson::JsonObject' has no member named 'prettyPrintTo'

   root.prettyPrintTo(Serial);

        ^
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

````

**_Library Versions used_**

1. Using library WiFiMQTTManager_Library at version 1.0.0-beta in folder
2. Using library WiFiManager at version 0.12 in folder
3. Using library ESP8266WiFi at version 1.0 in folder 
4. Using library ESP8266WebServer at version 1.0 in folder 
5. Using library DNSServer at version 1.1.0 in folder
6. Using library ArduinoJson at version 6.7.0-beta in folder
7. Using library PubSubClient at version 2.7 in folder

dreed47, 2018-12-17:
can you  please try the development branch of WiFiManager code https://github.com/tzapu/WiFiManager/tree/development

apanasara, 2018-12-18:
**_New Response_** after using suggested library

```
C:\Users\Amit Suani\Documents\Arduino\libraries\WiFiMQTTManager_Library\src\WiFiMQTTManager.cpp: In member function 'void WiFiMQTTManager::setup(String)':

C:\Users\Amit Suani\Documents\Arduino\libraries\WiFiMQTTManager_Library\src\WiFiMQTTManager.cpp:117:5: error: 'DynamicJsonBuffer' was not declared in this scope

     DynamicJsonBuffer jsonBuffer;

     ^

C:\Users\Amit Suani\Documents\Arduino\libraries\WiFiMQTTManager_Library\src\WiFiMQTTManager.cpp:117:23: error: expected ';' before 'jsonBuffer'

     DynamicJsonBuffer jsonBuffer;

                       ^

C:\Users\Amit Suani\Documents\Arduino\libraries\WiFiMQTTManager_Library\src\WiFiMQTTManager.cpp:118:24: error: 'jsonBuffer' was not declared in this scope

     JsonObject& json = jsonBuffer.createObject();

                        ^

C:\Users\Amit Suani\Documents\Arduino\libraries\WiFiMQTTManager_Library\src\WiFiMQTTManager.cpp:130:10: error: 'ArduinoJson::JsonObject' has no member named 'prettyPrintTo'

     json.prettyPrintTo(Serial);

          ^

C:\Users\Amit Suani\Documents\Arduino\libraries\WiFiMQTTManager_Library\src\WiFiMQTTManager.cpp:131:10: error: 'ArduinoJson::JsonObject' has no member named 'printTo'

     json.printTo(configFile);

          ^

C:\Users\Amit Suani\Documents\Arduino\libraries\WiFiMQTTManager_Library\src\WiFiMQTTManager.cpp: In member function 'void WiFiMQTTManager::_setupSpiffs()':

C:\Users\Amit Suani\Documents\Arduino\libraries\WiFiMQTTManager_Library\src\WiFiMQTTManager.cpp:192:9: error: 'DynamicJsonBuffer' was not declared in this scope

         DynamicJsonBuffer jsonBuffer;

         ^

C:\Users\Amit Suani\Documents\Arduino\libraries\WiFiMQTTManager_Library\src\WiFiMQTTManager.cpp:192:27: error: expected ';' before 'jsonBuffer'

         DynamicJsonBuffer jsonBuffer;

                           ^

C:\Users\Amit Suani\Documents\Arduino\libraries\WiFiMQTTManager_Library\src\WiFiMQTTManager.cpp:193:28: error: 'jsonBuffer' was not declared in this scope

         JsonObject& json = jsonBuffer.parseObject(buf.get());

                            ^

C:\Users\Amit Suani\Documents\Arduino\libraries\WiFiMQTTManager_Library\src\WiFiMQTTManager.cpp:194:14: error: 'ArduinoJson::JsonObject' has no member named 'printTo'

         json.printTo(Serial);

              ^

C:\Users\Amit Suani\Documents\Arduino\libraries\WiFiMQTTManager_Library\src\WiFiMQTTManager.cpp:195:18: error: 'ArduinoJson::JsonObject' has no member named 'success'

         if (json.success()) {

                  ^

C:\Users\Amit Suani\Documents\Arduino\libraries\WiFiMQTTManager_Library\src\WiFiMQTTManager.cpp: In member function 'void WiFiMQTTManager::_registerDevice()':

C:\Users\Amit Suani\Documents\Arduino\libraries\WiFiMQTTManager_Library\src\WiFiMQTTManager.cpp:266:3: error: 'StaticJsonBuffer' was not declared in this scope

   StaticJsonBuffer<2000> JSONbuffer;

   ^

C:\Users\Amit Suani\Documents\Arduino\libraries\WiFiMQTTManager_Library\src\WiFiMQTTManager.cpp:266:26: error: 'JSONbuffer' was not declared in this scope

   StaticJsonBuffer<2000> JSONbuffer;

                          ^

C:\Users\Amit Suani\Documents\Arduino\libraries\WiFiMQTTManager_Library\src\WiFiMQTTManager.cpp:282:8: error: 'ArduinoJson::JsonObject' has no member named 'printTo'

   root.printTo(messageBuffer, sizeof(messageBuffer));

        ^

C:\Users\Amit Suani\Documents\Arduino\libraries\WiFiMQTTManager_Library\src\WiFiMQTTManager.cpp:286:8: error: 'ArduinoJson::JsonObject' has no member named 'prettyPrintTo'

   root.prettyPrintTo(Serial);

        ^
```