dreed47/WifiMQTTManager #2: Example failing to compile

Danielv123, 2019-02-13:
C:\Users\danie\Documents\Arduino\libraries\WiFiMQTTManager_Library\src\WiFiMQTTManager.cpp: In member function 'void WiFiMQTTManager::setup(String)':

C:\Users\danie\Documents\Arduino\libraries\WiFiMQTTManager_Library\src\WiFiMQTTManager.cpp:56:42: error: 'WIFI_getChipId' was not declared in this scope

   String _chipId = String(WIFI_getChipId(),HEX);

                                          ^

C:\Users\danie\Documents\Arduino\libraries\WiFiMQTTManager_Library\src\WiFiMQTTManager.cpp:75:4: error: no matching function for call to 'WiFiManager::setSaveConfigCallback(WiFiMQTTManager::setup(String)::__lambda0)'

   });

    ^

C:\Users\danie\Documents\Arduino\libraries\WiFiMQTTManager_Library\src\WiFiMQTTManager.cpp:75:4: note: candidate is:

In file included from C:\Users\danie\Documents\Arduino\libraries\WiFiMQTTManager_Library\src\WiFiMQTTManager.h:12:0,

                 from C:\Users\danie\Documents\Arduino\libraries\WiFiMQTTManager_Library\src\WiFiMQTTManager.cpp:7:

C:\Users\danie\Documents\Arduino\libraries\WiFiManager/WiFiManager.h:109:19: note: void WiFiManager::setSaveConfigCallback(void (*)())

     void          setSaveConfigCallback( void (*func)(void) );

                   ^

C:\Users\danie\Documents\Arduino\libraries\WiFiManager/WiFiManager.h:109:19: note:   no known conversion for argument 1 from 'WiFiMQTTManager::setup(String)::__lambda0' to 'void (*)()'

C:\Users\danie\Documents\Arduino\libraries\WiFiMQTTManager_Library\src\WiFiMQTTManager.cpp:76:7: error: 'class WiFiManager' has no member named 'setClass'

   wm->setClass("invert"); // dark theme

       ^

C:\Users\danie\Documents\Arduino\libraries\WiFiMQTTManager_Library\src\WiFiMQTTManager.cpp:117:5: error: 'DynamicJsonBuffer' was not declared in this scope

     DynamicJsonBuffer jsonBuffer;

     ^

C:\Users\danie\Documents\Arduino\libraries\WiFiMQTTManager_Library\src\WiFiMQTTManager.cpp:117:23: error: expected ';' before 'jsonBuffer'

     DynamicJsonBuffer jsonBuffer;

                       ^

C:\Users\danie\Documents\Arduino\libraries\WiFiMQTTManager_Library\src\WiFiMQTTManager.cpp:118:24: error: 'jsonBuffer' was not declared in this scope

     JsonObject& json = jsonBuffer.createObject();

                        ^

C:\Users\danie\Documents\Arduino\libraries\WiFiMQTTManager_Library\src\WiFiMQTTManager.cpp:130:10: error: 'ArduinoJson::JsonObject' has no member named 'prettyPrintTo'

     json.prettyPrintTo(Serial);

          ^

C:\Users\danie\Documents\Arduino\libraries\WiFiMQTTManager_Library\src\WiFiMQTTManager.cpp:131:10: error: 'ArduinoJson::JsonObject' has no member named 'printTo'

     json.printTo(configFile);

          ^

C:\Users\danie\Documents\Arduino\libraries\WiFiMQTTManager_Library\src\WiFiMQTTManager.cpp: In member function 'void WiFiMQTTManager::_setupSpiffs()':

C:\Users\danie\Documents\Arduino\libraries\WiFiMQTTManager_Library\src\WiFiMQTTManager.cpp:192:9: error: 'DynamicJsonBuffer' was not declared in this scope

         DynamicJsonBuffer jsonBuffer;

         ^

C:\Users\danie\Documents\Arduino\libraries\WiFiMQTTManager_Library\src\WiFiMQTTManager.cpp:192:27: error: expected ';' before 'jsonBuffer'

         DynamicJsonBuffer jsonBuffer;

                           ^

C:\Users\danie\Documents\Arduino\libraries\WiFiMQTTManager_Library\src\WiFiMQTTManager.cpp:193:28: error: 'jsonBuffer' was not declared in this scope

         JsonObject& json = jsonBuffer.parseObject(buf.get());

                            ^

C:\Users\danie\Documents\Arduino\libraries\WiFiMQTTManager_Library\src\WiFiMQTTManager.cpp:194:14: error: 'ArduinoJson::JsonObject' has no member named 'printTo'

         json.printTo(Serial);

              ^

C:\Users\danie\Documents\Arduino\libraries\WiFiMQTTManager_Library\src\WiFiMQTTManager.cpp:195:18: error: 'ArduinoJson::JsonObject' has no member named 'success'

         if (json.success()) {

                  ^

C:\Users\danie\Documents\Arduino\libraries\WiFiMQTTManager_Library\src\WiFiMQTTManager.cpp: In member function 'void WiFiMQTTManager::_registerDevice()':

C:\Users\danie\Documents\Arduino\libraries\WiFiMQTTManager_Library\src\WiFiMQTTManager.cpp:266:3: error: 'StaticJsonBuffer' was not declared in this scope

   StaticJsonBuffer<2000> JSONbuffer;

   ^

C:\Users\danie\Documents\Arduino\libraries\WiFiMQTTManager_Library\src\WiFiMQTTManager.cpp:266:26: error: 'JSONbuffer' was not declared in this scope

   StaticJsonBuffer<2000> JSONbuffer;

                          ^

C:\Users\danie\Documents\Arduino\libraries\WiFiMQTTManager_Library\src\WiFiMQTTManager.cpp:282:8: error: 'ArduinoJson::JsonObject' has no member named 'printTo'

   root.printTo(messageBuffer, sizeof(messageBuffer));

        ^

C:\Users\danie\Documents\Arduino\libraries\WiFiMQTTManager_Library\src\WiFiMQTTManager.cpp:286:8: error: 'ArduinoJson::JsonObject' has no member named 'prettyPrintTo'

   root.prettyPrintTo(Serial);

        ^

Multiple libraries were found for "WiFiMQTTManager.h"
 Used: C:\Users\danie\Documents\Arduino\libraries\WiFiMQTTManager_Library
 Not used: C:\Users\danie\Documents\Arduino\libraries\arduino_228333
exit status 1
Error compiling for board NodeMCU 1.0 (ESP-12E Module).

dreed47, 2019-02-13:
can you please provide the version numbers you're using for the following:

- Arduino IDE
- WiFiManager Library (must use development branch)
- Arduino Json Library
- PubSub Client Library

Also, what specific board are you testing with?

Danielv123, 2019-02-14:
Ah, I wasn't using the development branch of WiFiManager. That explains this issue.

closed 2019-02-14 by Danielv123