dreed47/WifiMQTTManager #4: Not going into AP Mode

heyhogan, 2019-04-26:
First thanks for putting this together.  This is exactly the kind of project I wanted to replicate.   Unfortunately, I am having issues with the unit not going into AP mode.  I have a Wemos D1 R1 chip (8266).  I have uploaded the sketch with the wwm.formatFS = true.  (FYI there is a typo in Basic sketch,  wm.formatFS should be "wwm.formatFS".  See attached Serial Output.  
[Serial Monitor Output.txt](https://github.com/dreed47/WifiMQTTManager/files/3122812/Serial.Monitor.Output.txt)

dreed47, 2019-04-27:
Does it only happen with one device?  Have you other ESP32 or ESP8266 boards?

heyhogan, 2019-05-01:
I think I must have had  a bad device.  I was able to flash a new one.  I
got it to work.  Please confirm that once you setup WiFI the captive portal
is no longer available to reset MQTT settings.  Thanks!

Kevin

nonix, 2019-05-05:
Hi, not sure if this is related or not, but after selecting "format FS" option from service dashboard, I have to physically press the RST button to restart in AP mode. My board is same/similar Wemos D1 mini R1 chip 8266

Thank you,
 Norbert

closed 2019-11-27 by dreed47