dreed47/WifiMQTTManager #6: The subscriptionCallback overrides _subscriptionCallback making Dahsboard ineffective

nonix, 2019-05-11:
Hi
I am not a 100% sure, but I think only one subscriptionCallback  can be registered. Once I register subscriptionCallback in the sketch (Eample: Basic.ino), the Dashboard Services tool will become ineffective, since any mqtt message sent is trapped by sketch subscriptionCallback rather than WifiMQTTManager::_subscriptionCallback, which e.g.  triggers restart

Thank you,
 N.

koffing, 2019-05-13:
Yes, only one subscriptionCallback can be registered and the one that runs (in my sketch) is the one inside the library file not from my sketch. Even after I disabled the subscriptionCallback inside the library, the callback in my sketch still not working. I'll tinker with the library more

TinajaLabs, 2019-06-20:
I think I'm having a similar issue...  confusion about which `subscriptionCallback` is being called; the one in `basic`, or the one in `WiFiMQTTManager`.

In my case the confusion happens after my wemos d1 mini loses connection to the mqtt server.  You can see it sets up and calls the `subscriptionCallback` as defined in top level, basic.ino (I've renamed it, but for simplicity, I'll call this basic.ino).

As an aside, you can see I've added a new parameter called `mqtt_topic_loc`.  I'll be pre-pending that string to the topic as the location attributes (city/building/room) info for all my devices.

Here's the startup from terminal window:
```18:27:25.658 -> WiFiMQTTManager: Night Light
18:27:25.658 -> 
18:27:25.658 -> 
18:27:25.691 -> WMM: clientId: Tinaja_a020a6194f75
18:27:25.691 -> WMM: mounting FS...
18:27:25.758 -> WMM: mounted file system...
18:27:25.758 -> WMM: reading config file...
18:27:25.791 -> WMM: opened config file...
18:27:25.791 -> {"friendly_name":"Under bed night light","mqtt_topic_loc":"kensington/home/bedroom","mqtt_server":"tinaja-co","mqtt_port":"1883"}
18:27:25.791 -> WMM: parsed json...
18:27:25.791 -> WMM: lets try to connect to WiFi...
18:27:25.791 -> WMM: entering config mode...
18:27:25.791 -> WMM: 192.168.4.1
18:27:25.791 -> WMM: connect your device to WiFi SSID Tinaja_a020a6194f75 to configure WiFi and MQTT...
18:28:26.993 -> WMM: should save config...
18:28:28.025 -> WMM: connected to WiFi!!...
18:28:28.025 -> WMM: saving config...
18:28:28.025 -> {
18:28:28.025 ->   "friendly_name": "Under bed night light",
18:28:28.025 ->   "mqtt_topic_loc": "kensington/home/bedroom",
18:28:28.025 ->   "mqtt_server": "tinaja-co",
18:28:28.025 ->   "mqtt_port": "1883"
18:28:28.025 -> }WMM: 
18:28:28.025 -> WMM: local IP:192.168.1.113
18:28:28.025 -> WMM: _mqtt_server: tinaja-co
18:28:28.025 -> WMM: _mqtt_port: 1883
18:28:28.025 -> WMM: _mqtt_topic_loc: kensington/home/bedroom
18:28:28.025 -> WMM: clientId: Tinaja_a020a6194f75
18:28:28.025 -> WMM: attempting MQTT connection...mqtt connected...via setup...
# --------------------->>
18:28:30.390 -> subscribing to some topics:
18:28:30.390 ->   kensington/home/roomnight-light/wemosd1-1/Tinaja_a020a6194f75/output
18:28:30.390 -> WMM: subscribed to service...
# <<------------------------
18:28:30.390 -> Sending message to MQTT topic: deviceLog/Tinaja_a020a6194f75
18:28:30.426 -> {
18:28:30.426 ->   "deviceType": "ESP8266",
18:28:30.426 ->   "deviceId": "Tinaja_a020a6194f75",
18:28:30.426 ->   "name": "Under bed night light",
18:28:30.426 ->   "location": "kensington/home/bedroom",
18:28:30.426 ->   "chipId": "194f75",
18:28:30.426 ->   "sketchName": "tinaja-wemosd1mini-neopixel-nightlight-1.ino"
18:28:30.426 -> }
18:28:30.426 -> messageBuffer: {"deviceType":"ESP8266","deviceId":"Tinaja_a020a6194f75","name":"Under bed night light","location":"kensington/home/bedroom","chipId":"194f75","sketchName":"tinaja-wemosd1mini-neopixel-nightlight-1.ino"}
18:28:30.456 -> WMM: Success sending message to register the device...
18:28:32.255 ->  tinaja started and waiting...
18:30:50.338 -> WMM: subscriptionCallback called... top level
18:30:50.338 -> Message arrived on topic in top level: kensington/home/roomnight-light/wemosd1-1/Tinaja_a020a6194f75/output. Payload: 1
18:30:50.338 -> free heap: 41376
18:30:50.338 -> mqtt pub - LED State: 1
18:30:55.333 -> WMM: subscriptionCallback called... top level
18:30:55.333 -> Message arrived on topic in top level: kensington/home/roomnight-light/wemosd1-1/Tinaja_a020a6194f75/output. Payload: 0
18:30:55.333 -> free heap: 41376
18:30:55.333 -> mqtt pub - LED State: 0
```
In the above you can see it configured to access `subscriptionCallback` at the top level, ie., in basic.ino.  

The following is where it fails to connect to mqtt and the retries begin.
```05:16:12.228 -> Message arrived on topic in top level: kensington/home/roomnight-light/wemosd1-1/Tinaja_a020a6194f75/output. Payload: 1
05:16:12.228 -> free heap: 41688
05:16:12.228 -> mqtt pub - LED State: 1
05:16:17.197 -> WMM: subscriptionCallback called... top level
05:16:17.230 -> Message arrived on topic in top level: kensington/home/roomnight-light/wemosd1-1/Tinaja_a020a6194f75/output. Payload: 0
[... 8 lines cut ...]
05:17:02.996 -> WMM: attempting MQTT connection...mqtt connect failed, rc=-4 try again in 5 seconds...
```

Then when it gets connected again, it comes back on the _subscriptionCallback which is not the same as subscriptionCallback.  As a result, the subscription is received but there is no code within _subscriptionCallback to turn the leds on or off.

```05:35:20.160 -> WMM: attempting MQTT connection...mqtt connect failed, rc=-2 try again in 5 seconds...
05:35:30.164 -> WMM: attempting MQTT connection...mqtt connect failed, rc=-4 try again in 5 seconds...
05:35:50.562 -> WMM: attempting MQTT connection...mqtt connected...via reconnect loop...
05:35:53.494 -> subscribing to some topics:
05:35:53.494 ->   kensington/home/roomnight-light/wemosd1-1/Tinaja_a020a6194f75/output
05:35:53.527 -> WMM: subscribed to service...
05:36:15.562 -> WMM: attempting MQTT connection...mqtt connected...via reconnect loop...
# --------------------->>
05:36:17.062 -> subscribing to some topics:
05:36:17.062 ->   kensington/home/roomnight-light/wemosd1-1/Tinaja_a020a6194f75/output
05:36:17.062 -> WMM: subscribed to service...
05:36:37.262 -> WMM: _subscriptionCallback called...
# <<------------------------
05:36:37.295 -> WMM: Message arrived on topic in WifiMQTTManager: kensington/home/roomnight-light/wemosd1-1/Tinaja_a020a6194f75/output. Message: 0
05:36:52.264 -> WMM: _subscriptionCallback called...
05:36:52.298 -> WMM: Message arrived on topic in WifiMQTTManager: kensington/home/roomnight-light/wemosd1-1/Tinaja_a020a6194f75/output. Message: 1
```

Thanks for any tips,
Chris.