dreed47/WifiMQTTManager #8: Error compiling

mpaneth, 2019-06-24:
Help please as I am trying to compile Basic

I have loaded the following libraries
ArduinoJson: v6.11.1
pubsubclient: v2.7
WifiManager: Development branch
WifiMQTTManager: v1.0.2-beta

When I compile I get the following output

Arduino: 1.8.9 (Windows Store 1.8.21.0) (Windows 10), Board: "DOIT ESP32 DEVKIT V1, 80MHz, 921600, None"

C:\Users\Mike\Documents\Arduino\libraries\WifiMQTTManager\examples\Basic\Basic.ino:7:46: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]

 WiFiMQTTManager wmm(RESET_BUTTON, AP_PASSWORD);  // AP_PASSWORD is defined in the secrets.h file

                                              ^

C:\Users\Mike\Documents\Arduino\libraries\WifiMQTTManager\src\WiFiMQTTManager.cpp:117:11: error: DynamicJsonBuffer is a class from ArduinoJson 5. Please see arduinojson.org/upgrade to learn how to upgrade your program to ArduinoJson version 6

     DynamicJsonBuffer jsonBuffer;

           ^

C:\Users\Mike\Documents\Arduino\libraries\WifiMQTTManager\src\WiFiMQTTManager.cpp:192:11: error: DynamicJsonBuffer is a class from ArduinoJson 5. Please see arduinojson.org/upgrade to learn how to upgrade your program to ArduinoJson version 6

         DynamicJsonBuffer jsonBuffer;

           ^

C:\Users\Mike\Documents\Arduino\libraries\WifiMQTTManager\src\WiFiMQTTManager.cpp:266:11: error: StaticJsonBuffer is a class from ArduinoJson 5. Please see arduinojson.org/upgrade to learn how to upgrade your program to ArduinoJson version 6

   StaticJsonBuffer<2000> JSONbuffer;

           ^

C:\Users\Mike\Documents\Arduino\libraries\WifiMQTTManager\src\WiFiMQTTManager.cpp: In member function 'void WiFiMQTTManager::setup(String)':

C:\Users\Mike\Documents\Arduino\libraries\WifiMQTTManager\src\WiFiMQTTManager.cpp:117:23: error: 'jsonBuffer' was not declared in this scope

     DynamicJsonBuffer jsonBuffer;

                       ^

C:\Users\Mike\Documents\Arduino\libraries\WifiMQTTManager\src\WiFiMQTTManager.cpp:130:10: error: 'ArduinoJson::JsonObject {aka class ArduinoJson6111_00000::ObjectRef}' has no member named 'prettyPrintTo'

     json.prettyPrintTo(Serial);

          ^

C:\Users\Mike\Documents\Arduino\libraries\WifiMQTTManager\src\WiFiMQTTManager.cpp:131:10: error: 'ArduinoJson::JsonObject {aka class ArduinoJson6111_00000::ObjectRef}' has no member named 'printTo'

     json.printTo(configFile);

          ^

C:\Users\Mike\Documents\Arduino\libraries\WifiMQTTManager\src\WiFiMQTTManager.cpp: In member function 'void WiFiMQTTManager::_setupSpiffs()':

C:\Users\Mike\Documents\Arduino\libraries\WifiMQTTManager\src\WiFiMQTTManager.cpp:192:27: error: 'jsonBuffer' was not declared in this scope

         DynamicJsonBuffer jsonBuffer;

                           ^

C:\Users\Mike\Documents\Arduino\libraries\WifiMQTTManager\src\WiFiMQTTManager.cpp:194:14: error: 'ArduinoJson::JsonObject {aka class ArduinoJson6111_00000::ObjectRef}' has no member named 'printTo'

         json.printTo(Serial);

              ^

C:\Users\Mike\Documents\Arduino\libraries\WifiMQTTManager\src\WiFiMQTTManager.cpp:195:18: error: 'ArduinoJson::JsonObject {aka class ArduinoJson6111_00000::ObjectRef}' has no member named 'success'

         if (json.success()) {

                  ^

C:\Users\Mike\Documents\Arduino\libraries\WifiMQTTManager\src\WiFiMQTTManager.cpp: In member function 'void WiFiMQTTManager::_registerDevice()':

C:\Users\Mike\Documents\Arduino\libraries\WifiMQTTManager\src\WiFiMQTTManager.cpp:266:19: error: expected primary-expression before '<' token

   StaticJsonBuffer<2000> JSONbuffer;

                   ^

C:\Users\Mike\Documents\Arduino\libraries\WifiMQTTManager\src\WiFiMQTTManager.cpp:266:26: error: 'JSONbuffer' was not declared in this scope

   StaticJsonBuffer<2000> JSONbuffer;

                          ^

C:\Users\Mike\Documents\Arduino\libraries\WifiMQTTManager\src\WiFiMQTTManager.cpp:282:8: error: 'ArduinoJson::JsonObject {aka class ArduinoJson6111_00000::ObjectRef}' has no member named 'printTo'

   root.printTo(messageBuffer, sizeof(messageBuffer));

        ^

C:\Users\Mike\Documents\Arduino\libraries\WifiMQTTManager\src\WiFiMQTTManager.cpp:286:8: error: 'ArduinoJson::JsonObject {aka class ArduinoJson6111_00000::ObjectRef}' has no member named 'prettyPrintTo'

   root.prettyPrintTo(Serial);

        ^

Multiple libraries were found for "WiFi.h"
 Used: C:\Users\Mike\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.2\libraries\WiFi
 Not used: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.21.0_x86__mdqgnx93n4wtt\libraries\WiFi
exit status 1
Error compiling for board DOIT ESP32 DEVKIT V1.

This report would have more information with "Show verbose output during compilation" option enabled in File -> Preferences.

nonix, 2019-06-24:
Hi I think the Json library version needs to be 5.x take the latest on 5.x
Cheers,

N.

mpaneth, 2019-06-24:
Thanks for the super-fast response.  :-)

V5.3.15 compiles now with only 1 warning

C:\Users\Mike\AppData\Local\Temp\arduino_modified_sketch_728169\Basic.ino:7:46: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]

 WiFiMQTTManager wmm(RESET_BUTTON, AP_PASSWORD);  // AP_PASSWORD is defined in the secrets.h file

                                              ^

Sketch uses 848174 bytes (64%) of program storage space. Maximum is 1310720 bytes.
Global variables use 39376 bytes (12%) of dynamic memory, leaving 288304 bytes for local variables. Maximum is 327680 bytes.

TinajaLabs, 2019-06-25:
@mpaneth - I modified the line like this and it seemed to quiet the issue:

```WiFiMQTTManager wmm(RESET_BUTTON, (char *)AP_PASSWORD);```

mpaneth, 2019-06-26:
Thank you for your help.  I can now compile cleanly, but can't seem to get the app to start.  

There is no output on the serial interface and no WiFi AP.

Any ideas??

Here is my compile & load log

Sketch uses 848170 bytes (64%) of program storage space. Maximum is 1310720 bytes.
Global variables use 39376 bytes (12%) of dynamic memory, leaving 288304 bytes for local variables. Maximum is 327680 bytes.
esptool.py v2.6
Serial port COM12
Connecting........___
Chip is ESP32D0WDQ6 (revision 1)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
MAC: b4:e6:2d:f6:ce:bd
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 8192 bytes to 47...

Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 8192.1 kbit/s)...
Hash of data verified.
Compressed 15328 bytes to 9994...

Writing at 0x00001000... (100 %)
Wrote 15328 bytes (9994 compressed) at 0x00001000 in 0.1 seconds (effective 1021.9 kbit/s)...
Hash of data verified.
Compressed 848288 bytes to 477335...

Writing at 0x00010000... (3 %)
Writing at 0x00014000... (6 %)
Writing at 0x00018000... (10 %)
Writing at 0x0001c000... (13 %)
Writing at 0x00020000... (16 %)
Writing at 0x00024000... (20 %)
Writing at 0x00028000... (23 %)
Writing at 0x0002c000... (26 %)
Writing at 0x00030000... (30 %)
Writing at 0x00034000... (33 %)
Writing at 0x00038000... (36 %)
Writing at 0x0003c000... (40 %)
Writing at 0x00040000... (43 %)
Writing at 0x00044000... (46 %)
Writing at 0x00048000... (50 %)
Writing at 0x0004c000... (53 %)
Writing at 0x00050000... (56 %)
Writing at 0x00054000... (60 %)
Writing at 0x00058000... (63 %)
Writing at 0x0005c000... (66 %)
Writing at 0x00060000... (70 %)
Writing at 0x00064000... (73 %)
Writing at 0x00068000... (76 %)
Writing at 0x0006c000... (80 %)
Writing at 0x00070000... (83 %)
Writing at 0x00074000... (86 %)
Writing at 0x00078000... (90 %)
Writing at 0x0007c000... (93 %)
Writing at 0x00080000... (96 %)
Writing at 0x00084000... (100 %)
Wrote 848288 bytes (477335 compressed) at 0x00010000 in 6.7 seconds (effective 1012.0 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 144...

Writing at 0x00008000... (100 %)
Wrote 3072 bytes (144 compressed) at 0x00008000 in 0.0 seconds (effective 2730.6 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...